thorrak/brewpi-esp #8: Unable to create beer profile

rmcarlsson, 2017-06-16:
When trying to start a beer profile I get an error code stating some kind of IO error:

I/O error (2) updating profile: No such file or directory:

stderr log:

Apr 10 2017 23:25:05 Setting profile 'trripel' as active profile
I/O Error(2) updating profile: No such file or directory 
Apr 10 2017 23:25:07 Setting profile 'trripel' as active profile
I/O Error(2) updating profile: No such file or directory 
Apr 10 2017 23:25:32 Setting profile 'trripel' as active profile
I/O Error(2) updating profile: No such file or directory 
Apr 10 2017 23:25:35 Setting profile 'trripel' as active profile
I/O Error(2) updating profile: No such file or directory 
Apr 10 2017 23:25:35 Setting profile 'trripel' as active profile
I/O Error(2) updating profile: No such file or directory 
Apr 10 2017 23:25:52 Setting profile 'Sample Profile' as active profile
I/O Error(2) updating profile: No such file or directory

rmcarlsson, 2017-06-18:
Could this be an installition script issue.  

Are the permissions correct?

pi@raspberrypi:/home/brewpi/settings $ ls -all  /var/www/data
total 32
drwxrwsrwx 8 brewpi   www-data 4096 Apr 12 15:00 .
drwxrwsr-x 4 www-data www-data 4096 Apr 10 23:22 ..
drwxrwxr-x 2 brewpi   www-data 4096 Apr 12 14:57 ddfdf
drwxrwxr-x 2 brewpi   www-data 4096 Apr 12 14:59 mikaels2
drwxrwxr-x 2 brewpi   www-data 4096 Apr 12 15:18 mikkkk
drwxrwxrwx 2 brewpi   www-data 4096 Apr 11 00:01 My First BrewPi Run
drwxr-sr-x 2 pi       www-data 4096 Apr 12 15:21 profiles
drwxrwxr-x 2 brewpi   www-data 4096 Apr 11 10:44 tripel
pi@raspberrypi:/home/brewpi/settings $ chown -R brewpi:www-data /var/www/data/

thorrak, 2017-06-19:
The installation script that handles file permissions shouldn't have been modified, but I'll need to do a fresh install to test. Let me look into this and I'll come back.

thorrak, 2019-12-07:
Closing this out, as all focus is now on either [Fermentrack](http://www.fermentrack.com/) or [BrewPiRemix](https://www.brewpiremix.com/) support.

closed 2019-12-07 by thorrak