thorrak/brewpi-esp #9: Error message when stopping the script

rmcarlsson, 2017-06-18:
Whn I press the "stop scrip" button in the web-gui I get an error message in the srderr log:

Traceback (most recent call last):
File "/home/brewpi/brewpi.py", line 690, in 
dontrunfile = open(dontRunFilePath, "w")
NameError: name 'dontRunFilePath' is not defined

rmcarlsson, 2017-06-18:
It looks like it is not possible to stop a brew. 

I have used the legacy branch in Brewpi before and this clone appears to be broken, at least if you would like to use fermentation profiles.

Let me know if need any more logs or data.

thorrak, 2017-06-19:
I'm guessing something must hav ebeen damaged when I was doing the Fermentrack conversion of brewpi-script. When I'm working on testing to determine the root cause of #8, I'll see about resolving this as well. Sorry about that!

thorrak, 2019-12-07:
Closing this out, as all focus is now on either [Fermentrack](http://www.fermentrack.com/) or [BrewPiRemix](https://www.brewpiremix.com/) support.

closed 2019-12-07 by thorrak