thorrak/brewpi-esp #12: Check router reconnection code

thorrak, 2017-09-23:
Users are reporting issues with ESP8266-based BrewPi installations not automatically reconnecting to routers when the connection is lost. This needs to be checked.

[Example here](https://github.com/thorrak/fermentrack/issues/202)

lggrizendi, 2017-10-18:
I have  the same issue using this https://github.com/letscontrolit/ESPEasy 
I don't if would help you but there is a open issue there to.
https://github.com/letscontrolit/ESPEasy/issues/413

thorrak, 2017-10-18:
This code right here was intended to correct the issue - no idea why it isn't working out that way:
https://github.com/thorrak/brewpi-esp8266/blob/master/legacy-platformio/src/brewpi-esp8266.cpp#L205

I haven't experienced the same issue on either of the two controllers I've had running for the past few months - but admittedly, both have line of sight to the router. I went ahead and picked up a spare $10 router for testing the other day. We'll see if that can be used to help test.

benbroadley88, 2018-03-15:
Hey @thorrak - was there ever any progress made in investigating this?

I'm encountering this issue (or something similar) frequently; At least once yesterday night and twice in the last few hours; I've not managed to brew a batch of beer without it cropping up at least once so far. 

If there's any device logs etc that would help to get to the bottom of this, then I'd be happy do to help.

The stderr log from Fermentrack shows the following but if I'm not mistaken, this is from the Raspberry Pi side not the ESP8266 so I guess of little use?:
``` Mar 15 2018 12:08:58   Error: controller is not responding to new data requests. Exiting.
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 754, in run
  File "/home/fermentrack/fermentrack/brewpi-script/backgroundserial.py", line 100, in __listenThread
  File "/home/fermentrack/fermentrack/brewpi-script/tcpSerial.py", line 44, in read
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'timeout'
 Mar 15 2018 12:09:32   Connection type WiFi selected.  Trying TCP serial (WiFi)
 Mar 15 2018 12:09:32   Connecting to BrewPi esp9039308.local (via 192.168.0.2) on port 23
 Mar 15 2018 12:09:42   Unable to connect to BrewPi 192.168.0.2 on port 23. Exiting.```

closed 2018-03-25 by thorrak

lggrizendi, 2018-03-26:
Every thing is working with this new version (v0.10) the ep8266 automatically  reconnect now. 
Thanks for the update.

commit 5ac186da (2018-03-25): Add WiFi reconnection fix (Closes #12)