thorrak/brewpi-esp #20: AP mode + some compatibility issues

brendongough, 2018-07-20:
Using a Wemos D1 mini pro (ESP8266EX) I seem to have issues connecting properly when in AP mode ie: first configuration. This is on multiple devices iOS / Android / OSX / Windows. After a lot of tries it does eventually work and I can configure the wifi client mode just fine.

I've also noticed different wifi routers work well and some have stability issues, this may point to the above problem. An AirPort Extreme AC works fine, the devie stays connected indefinitely where as my Mikrotrik devices come back generally with "Unicast key exchange timeout" issues with the wifi. Iv'e spent a bit of time trying to diagnose the wifi settings, using only low speed options, low freq channels and fixing channels, amongst other things. I eventually gave up and just use the Apple device. Confirming I am also using these with an external antenna as well as varied distances from the APs.

To rule out hardware issues, I've tried the same ESP8266EXs and moved them into an iSpindel board. With iSpindel I've noticed no issues connecting to the AP for initial config and no stability issues with any wifi router/AP in the house.

My question then is, the issue seems to be related to the software wifi stack being used. Are there any large differences between the one used in the iSpindel and that used in either brewpi-esp8266 and brewpiless, both have issues.

thorrak, 2018-07-20:
The long delay on first boot for Mini Pro devices is probably due to the fact that when SPIFFS (the flash filesystem) first is initialized, if it's on a brand new device it formats the flash. Mini Pro's have loads of flash, hence the delay. I've heard it can take 5 mins or longer in some cases. 

That said - do you have an LCD connected to your Mini Pros? If so, what does the LCD say? This is a common enough complaint, that I thought I had added a message that would appear on first boot to explain what was occuring. I'm wondering if that message might not be working properly.

There is no difference in the WiFi stack between BrewPi-ESP8266, BrewPiLess, and iSpindel. All of them use the default WiFi stack provided by the ESP8266 libraries. TBQH, I don't think there are alternative WiFi stacks available out there - I think everyone uses the one provided by Espressif!

My guess on why you're having issues with the Mikrotrik devices is that BrewPi-ESP8266 (and possibly BrewPiLess - haven't checked the code to verify) uses mDNS (Bonjour/Avahi) to announce itself for automatic configuration with Fermentrack. A small number of routers seem to have weirdness around mDNS support for some reason -- no idea why. 

If you can find a way to test/verify that mDNS is/is not causing the issue I'd be curious to know what you find. If you aren't, I can take a stab at creating a custom firmware version without the mDNS code this weekend for you to test/play with. Might not have an impact - but it also might fix the issue. Would be interesting either way!

brendongough, 2018-08-13:
Still happy to test this when ready :)

thorrak, 2018-08-13:
Ack - thanks for the reminder. Been in and out of town the past 4 weeks, and this slipped my mind. Let me pull this together and make it available.

thorrak, 2018-08-14:
This firmware has been uploaded and is available through Fermentrack. The name is "BrewPi /w WiFi - BETA FIRMWARE". The only change in this firmware is to disable calls to mDNS -- otherwise it's identical to the other (released) firmware.

Although calls to actually use mDNS are disabled, it still asks for mDNS name and whatnot. Just ignore it/set it however you like.

brendongough, 2018-08-18:
Looking good so far, I'll do some more compatibility testing and report back.

Massive thanks :)

thorrak, 2022-03-24:
Unfortunately, I've managed to leave this issue open for the past three and a half years. Sorry about that. :(

There has long been the bug in v0.11 where the AP continues to run after the device is configured. It's annoying - even if harmless. That bug has been fixed in the `experimental` branch, which contains a fair bit of code I'd like to release in the near future.

This branch also contains substantial rework of the WiFi & mDNS  code more broadly, which I believe should resolve the concerns that were raised. I'm going to tag this issue as "pending pull" and will close it when the code is merged. If there are further WiFi issues in the future, we can open a new issue to track/resolve.

commit 5c4631b7 (2022-04-06): Fix persistent AP bug (Fixes #20)
closed 2023-03-26 by thorrak