thorrak/brewpi-esp #24: ESP8266 BrewPi RJ-11 Sensor Breakout v1.1 eagle missing from hardware section

arnljot, 2019-03-29:
I've ordered this from PCBs.io but they are taking for ever and have no contact info. So I was planning to order from OshPark.

But it seems that ESP8266 BrewPi RJ-11 Sensor Breakout v1.1 eagles are not publicly available. Oversight or by design?

I hope that all PCBs can be sourced somewhere faster and more customer friendly than a site with no means of contacting them.

thorrak, 2019-03-29:
I actually just ordered some stuff from them which shipped today - hopefully you’re in the same batch. If you haven’t heard anything try reaching out to support@pcbs.io 

Lack of eagle files is certainly not by design. I’ll try to get everything uploaded this coming weekend so you can try another fab house if they aren’t coming through.

lbussy, 2019-03-30:
Eagle saves as a .brd file which I see in the hardware directory.

I received my boards from DirtyPCB in no time ... three weeks which is crazy fast from China.

thorrak, 2019-12-07:
I went through last weekend and updated all of the hardware designs, all of which are now part of this repo. I'm planning at a future date to move the hardware designs out into their own repo as well, which hopefully will help with organization.

Since this board is no longer a recommended board (everything is now RJ-45) I'm closing the issue.

closed 2019-12-07 by thorrak