thorrak/brewpi-esp #28: LCD showing only 1st character of each item

art-h, 2019-07-30:
Compiling the master branch using Visual Studio Code & PlatformIO generates code that only displays the first character of each bit of text on the LCD screen. 

Something like:

M        F         C
B        -       -   °
F        7       7   °
I i              5  

instead of:

Mode   Fridge Const.
Beer   --.-  --.- °F
Fridge 71.7  73.0 °F
Idling for   5h21m50

It appears the IIClcd.print function in the current set of platform code and supporting packages might have some issues. I've duct-taped together a fix via lilcd.h, lilcd.cpp, and DisplayLCD.cpp, but it's not exactly elegant.  Here's the build info, in case there's something obvious in the configuration that might fix it:

Processing esp8266_wifi (platform: espressif8266; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 2.2.2 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0), framework-arduinoespressif8266 2.20502.0 (2.5.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WifiManager> 1.0.0 #4775b1d
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
Linking .pio/build/esp8266_wifi/firmware.elf
Retrieving maximum program size .pio/build/esp8266_wifi/firmware.elf
Checking size .pio/build/esp8266_wifi/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  38.4% (used 31420 bytes from 81920 bytes)
PROGRAM: [====      ]  39.8% (used 415744 bytes from 1044464 bytes)
============================================== [SUCCESS] Took 3.22 seconds ==============================================

I'll be glad to check in my code if it's useful to help fix the issue, but I'm hoping someone that is more familiar with the code can give it a proper fix.

- Art

art-h, 2019-07-30:
Ah - one more clarification - I'm referring to the LCD view on Fermentrack. I don't have a physical LCD attached to my ESP8622 module.

zrgravity, 2019-08-22:
Do you use the LCD? I was able to get the same working by commenting out the LCD definitions in Config.h and DefaultConfig.h

mp-se, 2019-12-07:
Its a change in the underlaying arduino libraries that causes this. I have submitted a fix that has not been merged yet

thorrak, 2019-12-07:
As @mp-se noted, I haven't yet merged the change in. The existing binaries should work, but new compilations need the changes that he submitted the pull request to make. I have been focused on an alternative implementation for the past few months and haven't had a chance to test the change on newly-compiled firmware but this change will get merged as soon as I can get it tested.

Thanks again @mp-se - sorry for the delay on this!

art-h, 2020-01-30:
Hi @thorrak - have you been able to make this merge and test it yet? Alternatively, how's the alternate implementation coming along? Thanks!

mp-se, 2020-01-31:
Use this for now, it has the LCD issue fixed... or you can fix it yourself, just one character to change ... https://github.com/mp-se/brewpi-esp8266

PoulKlemmensen, 2021-01-31:
> Use this for now, it has the LCD issue fixed... or you can fix it yourself, just one character to change ... https://github.com/mp-se/brewpi-esp8266

Thanks. I have just been trying to compile the code myself and ran into the same issue. Want to change some constants and maybe add fan control if I can.

speshak, 2021-01-31:
@PoulKlemmensen You may want to work off of the esp32 branch.  Despite the name, it has a load of refactoring work not related to esp32 merged into it that might make your experience a little better.

thorrak, 2021-02-01:
@PoulKlemmensen as @speshak suggested, I highly recommend any new work being done off the esp32 branch. (Plus, if you want the future goodies that will eventually come with the esp32 branch, it will be a lot easier to pull them into your fork ;) )

PoulKlemmensen, 2021-02-01:
Hi.
Sorry for not doing my homework properly and thanks for pointing this out to me?
When you are talking about the ESP32 brach, are you then referring to Brewpi remix?

thorrak, 2021-02-01:
> Hi.
> Sorry for not doing my homework properly and thanks for pointing this out to me?
> When you are talking about the ESP32 brach, are you then referring to Brewpi remix?

There’s no documentation on this specific choice, so there’s no way you could have known. When we are referring to the ESP32 branch, we mean “branch” in the Git context. Specifically, we mean this: https://github.com/thorrak/brewpi-esp8266/tree/esp32

The BrewPi-Remix firmware is specifically for the Arduino Uno, not an ESP8266/ESP32.

closed 2023-03-26 by thorrak