thorrak/brewpi-esp #29: Incorrect level shift on LCD TH Screws and Dupont schematic

wvolz, 2019-08-21:
The LCD TH and Dupont schematics incorrectly have the SDA I2C line connected to a one way voltage divider for conversion from 5V to 3.3V (in red box below). The fix is to either move to a different level converter board or to move the D2 connection to the second bi-directional converter on the level shifter board (in green below).

[image]

It looks like the schematic was built around [this level converter](https://www.sparkfun.com/products/retired/11978), but [there is a version](https://www.sparkfun.com/products/12009) with 4 bi-directional shifters that should work as a drop-in replacement.

The schematic/board should be updated to show the correct level converter. I can submit a revised board+schematic if that would help.

thorrak, 2019-08-21:
If you don’t mind, that would be very helpful!

[20190821 brewpi-esp8266 th screws]<https://user-images.githubusercontent.com/404202/63472400-3b5d7b80-c42f-11e9-820e-5e1dfdf76c5c.png>

lalo-uy, 2019-08-22:
There is a quick hack that allow the I2C interface for the LCD to work on
3,3V and the LCD at 5V

https://www.instructables.com/id/Raspberry-Pi-Using-1-I2C-LCD-Backpacks-for-1602-Sc/

No need for level translators

wvolz, 2019-08-23:
@thorrak pull request submitted with fix. Attached is a screenshot of the change.

[image]

[image]

wvolz, 2019-08-23:
@lalo-uy You are correct that changing the I2C backpack board would eliminate the need for level shifters.

thorrak, 2019-12-07:
As @wvolz noted, the module included was incorrect. I've updated all of the PCBs since so hopefully this is now resolved.

Apologies for the delay in getting this out there!

closed 2019-12-07 by thorrak