thorrak/brewpi-esp #35: Failed to write the configuration to the controller.

Antti1111, 2019-12-25:
Hi
I have fermentrack and I can't save pin settings or temp sensors. I get "Failed to write the configuration to the controller. If this continues, try resetting the EEPROM on the controller". 

I try EEPROM resetting button, no help
I try reflash device, no help
Wemos d1 mini 8266 4M is device, firmware v0.11 wifi

Dec 25 2019 19:20:53   Received applyDevice request, updating to: {"i": 0, "c": 1, "b": 0, "f": 2, "h": 1, "p": 16, "x": "1"}
 Dec 25 2019 19:20:56   Device updated to: {"i":0,"t":0,"c":-26,"b":30,"f":72,"h":0,"d":0,"p":3}
 Dec 25 2019 19:20:56   Controller debug message: ERROR 7: Invalid config for device owner type 2 beer=0 chamber=1
 Dec 25 2019 19:20:56   Installed devices received: []
 Dec 25 2019 19:20:56   Controller debug message: ERROR 3: Device definition update specification is invalid
 Dec 25 2019 19:20:56   Available devices received: [{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 16, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 14, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 13, "x": 1}]

What next?

Antti1111, 2019-12-26:
Now it works, I don't know what happend 😀

closed 2019-12-26 by Antti1111

korycins, 2020-04-11:
I had the same issue. The solution for me was assigning a `Control Mode` to Device.