thorrak/brewpi-esp #37: Stuck wait to heat

Kooza25, 2020-02-10:
I set up a second room, just a very small room with an esp8266, a dual relay but just one heater to it. Of course just the heater port is set up within fermentrack.
When needed, I can see wait to heat and some minutes countdown, but when it reaches 0 it goes to “cannot receive info” and so on, then it seems to reboot the script and back again to wait to heat

thorrak, 2020-02-10:
It's definitely a firmware issue then. Transferring the issue over to the brewpi-esp8266 repo and I'll take a look.

For the time being, my guess is that the issue relates to the cooling pin not being assigned. It might make sense to assign it to something, even though you won't actually be using it to control anything.

Kooza25, 2020-02-10:
How could I reset completely the esp, to setup again wifi connection and flash it? Just to tryout

thorrak, 2020-02-10:
Either use BrewFlasher and choose the erase flash option, or flash the wiring test firmware and let it run through everything. The last thing it does is reset the WiFi and reinitialize SPIFFS

Kooza25, 2020-02-11:
Alright I’ll try that. The update for esp and uno devices is a manual process or the system does it by itself?

closed 2023-03-26 by thorrak