thorrak/brewpi-esp #39: Issue ordering at pcbs.io, where to find pcb layout

vihar17, 2020-05-21:
Hello,
I have issues ordering the boards at pcbs.io
Seems like they stopped working... (https://github.com/hallard/WeMos-Matrix-Shield/issues/6)

Is it possible to obtain the layout of the boards, so it's possible to order them elsewhere ?

Thank you

thorrak, 2020-05-21:
Just FYI - All of the PCBs in my repo are old. Let me know what PCBs you want, and I'll attach them here. The repo will get updated in the near future, but I need to fix a few PCBs first.

vihar17, 2020-05-21:
Thank you for your answer.

I'm looking for the D1 Mini Mosfet board (shown here : https://github.com/brewpi-remix/brewpi-pcb-rmx/tree/master/D1%20Breakout)
And the RJ45 board (shown here : https://github.com/thorrak/brewpi-esp8266/blob/master/hardware/BrewPi%20Sensor%20Boards/RJ-45%20Sensor%20Breakout.md)

thorrak, 2020-05-21:
The D1 mini mosfet board is @lbussy 's creation - I think he does a better job than I do keeping his files up to date, so I'm guessing that one is OK. 

The RJ45 board shouldn't ahve changed much, but I believe the latest design is this one:

[RJ-45 Top Routed.zip](https://github.com/thorrak/brewpi-esp8266/files/4665162/RJ-45.Top.Routed.zip)

vihar17, 2020-05-21:
Thank you very much !

closed 2020-05-21 by vihar17

lbussy, 2020-05-22:
The [D1 TH Mosfet board](https://github.com/brewpi-remix/brewpi-pcb-rmx/tree/master/ESP8266%20Boards/WeMos%20D1%20Mini%20Breakout) is mine, [the corresponding breakout](https://github.com/brewpi-remix/brewpi-pcb-rmx/tree/master/ESP8266%20Boards/RJ45%20Breakout) is this one.  There was a small error on the silk screening only on that board, which reversed the pinout.  Electrically it was correct and matched @thorrak 's specification.  I corrected the silkscreen this morning.

Being early in the morning, I have not looked to see if @thorrak has a TH MOSFET board.  I suspect not since he and I have "artistic differences."  :stuck_out_tongue_winking_eye:  Pinout-wise, my boards are interchangeable.

(BTW:  I moved the folder structure on mine this morning to make it a little less confusing.)

vihar17, 2020-05-22:
Oh crap I ordered another pcb...
May I ask why the GND isn't linked to anything on your board ?
I thought it was a mistake, which is why I ordered this board : https://aisler.net/p/SDEFMNWJ
Will it be compatible with your Mosfet board ?

Sorry for the aisler link, I don't remember where I found it originally

lbussy, 2020-05-22:
> May I ask why the GND isn't linked to anything on your board ?

To whom are you addressing that question?

thorrak, 2020-05-22:
> > May I ask why the GND isn't linked to anything on your board ?
> 
> To whom are you addressing that question?

I think me, because I think that board had a ground plane on the back. At least, I _hope_ it had a ground plane on the back.

vihar17, 2020-05-22:
> [the corresponding breakout](https://github.com/brewpi-remix/brewpi-pcb-rmx/tree/master/ESP8266%20Boards/RJ45%20Breakout) is this one

I'm talking about this board. It seems like GND isn't connected to anything (see here : https://aisler.net/p/BJRXNPCS)

lbussy, 2020-05-22:
What "issue"?  I assure you it is connected, at least when I open it (and had them manufactured.)

vihar17, 2020-05-22:
Thank you for the information.
My mistake.