thorrak/brewpi-esp #40: LANdesk Remote Management

HuggableShark, 2020-05-22:
I wanted to confirm if this is a legitimate service needed by the ESP module for updates from Fermentrack. Ran across it while performing a home network test and obviously it seemed dicey at first glance.

thorrak, 2020-05-22:
Where are you seeing it/on what device?

lbussy, 2020-05-22:
LANDesk uses a bunch of very common ports.  It would be more helpful if you posted the exact traffic you are seeing:  ports, directionality, protocol.

Also, the name of the tool you are using would help.  I suspect it's trying to be "helpful" without really knowing what it's found.

HuggableShark, 2020-05-23:
I was just running a nmap -A scan to check for any strange ports that may have opened. It was listening for TCP on port 23. Service is landesk-rc. I haven't performed any traffic sniffing on it yet, I just found the open port and a listening remote management app strange so I wanted to confirm if it was intended.

lbussy, 2020-05-23:
23 (TCP) is Telnet and I'm pretty sure how Fermentrack connects to the controller.

That's the problem with runing random tools - they lack context.

thorrak, 2020-05-23:
> 23 (TCP) is Telnet and I'm pretty sure how Fermentrack connects to the controller.
> 
> That's the problem with runing random tools - they lack context.

^^ This. 

@HuggableShark If you're curious, you can telnet into your controller. Type a single "l" (and hit enter maybe?) and you'll see it spit out your LCD info. If you're really curious about what the commands are, look here:
https://github.com/thorrak/brewpi-esp8266/blob/master/src/PiLink.cpp#L213

HuggableShark, 2020-05-23:
Appreciate the info. I'll probably do that for curiosity sake. Thank you both for the quick response!

closed 2020-05-23 by HuggableShark

lbussy, 2020-05-23:
> If you're really curious about what the commands are, look here:

.. or if code makes your eyes cross, I believe all of the original commands will work.  They are [documented here](
https://brewpi-userdocs-rmx.readthedocs.io/en/latest/api/index.html) (under "Arduino Commands").

thorrak, 2020-05-23:
I added a few! But yes, @lbussy 's documentation should be accurate(ish)