thorrak/brewpi-esp #42: [Feature Request] NTC temperature sensor support using A0

hefevice, 2020-05-27:
I am in the process of designing Brewpi-ish setup to replace the woeful temperature controller that comes the Grainfather Conical Pro (there has been talk of a WiFi based replacement from Grainfather but I can't find it for sale anywhere). I am taking the "Brain Transplant" approach where I will simply swap the controller board for a customised Brewpi D1 Mini based board which conveniently connects via ribbon cable to the existing interface circuitry. I have reverse engineered the connections and integration to BrewPi will be fairly straightforward except for 2 issues; Glycol cooling (minor) and the temperature probe appears to be NTC (and quite difficult to replace with a DS18x20).

As I have been using Fermentrack to manage my iSpindel devices, I am leaning towards using brewpi-esp8266 as a base rather than brewpiless (assuming the Fermentrack integration will be smoother).

I am in the early phase of getting my head around the BrewPi temperature configuration, sensing and control code, which appears to be very attuned to the Onewire devices. I would be happy to have a go it implementing this on a branch if it is considered useful and feasible and would appreciate any tips on how best to approach integrating this into the code. I understand that the analog input is also being used for pressure sensing; there would need to be some configuration to manage choosing one or the other as well.

Tuning Glycol will be phase 2. It should be fairly straightforward with the Grainfather as the chiller looks after control of the Glycol temperature, all the brewpi needs to do is switch the pump on and off.

thorrak, 2020-05-28:
Unfortunately, I know absolutely nothing about NTC probes, so I can't help building out that part - but I think implementing this should be relatively straightforward. 

The temperature gets read by this line, which reads the temperature via an abstracted class:
https://github.com/thorrak/brewpi-esp8266/blob/1c96f818b26b0b9b3b809a0cafa9930262ecd3a2/src/TempSensor.cpp#L44

To implement NTC temperature sensor support, you would just need to build out a new temp sensor class based on the OneWire one:
https://github.com/thorrak/brewpi-esp8266/blob/json_config/src/OneWireTempSensor.cpp

As long as you implement the appropriate methods, it should work. You'll then need to modify this line to allow for adding an NTC temp sensor rather than an OneWire one and you'd be set, I think:
https://github.com/thorrak/brewpi-esp8266/blob/1c96f818b26b0b9b3b809a0cafa9930262ecd3a2/src/DeviceManager.cpp#L127

lalo-uy, 2020-05-28:
The code to read the NTC can be borrowed fron stc1000+ ( open firmware for
stc1000 controler ).
I managed to emulate a DS18B20 with a Pic12f1572 reading a NTC with it.

closed 2022-03-24 by thorrak