thorrak/brewpi-esp #44: Firmware Not Detecting DS18B20 Tempriture sensors

simonboydfoley, 2020-06-28:
I've been pulling my hair out with this firmware, i have tried all flavours of the ESP8266 for my wemos  D1 Mini ESP8266MOD without success. I have tested the circuity and tested the sensors seperatly on the same circuit with a Mega 2560 and Arduino Nano using the oneWire and Dallas libraries and the circuit tests good with the tempriture sensors. so its this firware that is broken. I have a scope on the oneWire data pin D6 and I can see the data preamble (although I have not ran my logic analyser on it yet). I just cant the get the tempriture of the sensors to display in fermentrack or the the LCD Display.

I'll go back and reread the documentation in case I missed somethng ...  

This is the error message I get;

Jun 28 2020 20:16:31   Connection type WiFi selected.  Trying TCP serial (WiFi)
 Jun 28 2020 20:16:31   Connecting to BrewPi brewpiwifi.local (via 192.168.100.116) on port 23
 Jun 28 2020 20:16:34   Successfully connected to controller.
 Jun 28 2020 20:16:34   Notification: Script started, with no active beer being logged
 Jun 28 2020 20:16:44   Checking software version on controller... 
 Jun 28 2020 20:16:44   Found BrewPi v0.2.4, running commit 00000000, running on an ESP8266 ESP8266 with a DIY shield on port 192.168.100.116:23

 Jun 28 2020 20:16:44   BrewPi version received was 0.2.4 which this script supports in 'legacy' branch mode.
 Jun 28 2020 20:16:44   Bound to TCP socket on port 2563, interface localhost 
 Jun 28 2020 20:16:45   Installed devices received: []
 Jun 28 2020 20:16:50   Available devices received: [{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": 0, "p": 12, "v": 23.063, "a": "283804955E14012B", "j": 0.0}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": 0, "p": 12, "v": 22.688, "a": "281AD7743214013B", "j": 0.0}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 16, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 14, "x": 1}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": 0, "p": 13, "x": 1}]
 Jun 28 2020 20:17:30   Notification: Beer temperature set to 23.0 degrees in web interface

thorrak, 2020-06-28:
From the log you posted, your controller is seeing two temperature sensors, both of which are reporting temperatures (presumably in deg C):
{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": 0, "p": 12, "v": 23.063, "a": "283804955E14012B", "j": 0.0}, 
{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": 0, "p": 12, "v": 22.688, "a": "281AD7743214013B", "j": 0.0}

(The "a" item is the OneWire address, and the "v" item is the value as read off the sensor)

That said, you currently don't have any functions assigned to any of your pins/sensors. What are you seeing when you go to the configure pins/sensors page in Fermentrack?

simonboydfoley1, 2020-06-29:
@thorrak hi John
You have just provided me the vital piece of info.
I've been wasting 1 week looking for a sensor issue / wiring issue.
I assumed that ferment track would auto display the temperature sensors it found like it auto configured the wifi.
That fermentrack pin configuration screen I could not find when reading the docs. I could not see it anywhere!
I assumed I could not see it because the sensors could not be detected.
I assumed it would be under the device configuration screen not some hidden away menu that's bloody hard to find.
After clicking around like mad I eventually found the pin config menu.
I bet I am not the only person caught out by this ....the pin config screen is in a really bad location and should be moved to the device config screen... We logically most people would look.
It confused the he'll out of people and wasted 1 week of my time even when reading fermentrack docs it's unclear where that pin config screen is!
Thanks for all your help. 
Turns out I never had a problem and wasted a week testing everything  :-(  
all user error And not ringing that pin config screen.
It should be moved.

closed 2022-03-23 by thorrak