thorrak/brewpi-esp #45: Add ability to name & log an arbitrary number of temperature probes

speshak, 2020-08-31:
I have additional temp probes that aren't one of the beer/fridge/room options. I'm using fermentrack/brewpi-8266 to control glycol chill loops on my fermenters.  I'm ok with the control probes not have "correct" names (for example, my fermenter probe is configured as the fridge probe in fermentrack), but I'd like to at least be able to log the data from these extra probes.  

I personally have a probe for the coolant bath and for the post-jacket coolant temp, but I think any solution should allow arbitrary names to be set on these non-control probes.

thorrak, 2020-08-31:
> The additional probes are just more probes on the OneWire bus. I am using a custom PCB based off yours that has additional probe connectors.
> 
> The probes show up fine as uninitiated devices, I just want to get their data into the CSV.

Ah, makes sense. This is a controller thing more than a Fermentrack one (for now) - this can be done but requires additional work. I’m moving this issue to the controller issue board for tracking.

speshak, 2020-08-31:
👍  I wasn't entirely sure where the bulk of the changes would have to go.  I'm more than happy to try to take a stab at the change.  Do you have any opinions on what the change would look like?

speshak, 2020-09-05:
I made a first stab at getting all of the temp probes into the `t` output: https://github.com/speshak/brewpi-esp8266/commit/5e4d8e2161c223da99e60c01241d8f9be340d0e7
This extends the existing JSON response with a `rawDevices` key which points to an array of sensor information.  An example from my test setup:

```"rawDevices": [{"device": "28E4D922292001B4", "value":  23.375},{"device": "28CAD17C292001E2", "value":  23.250},{"device": "285B4E17292001B8", "value":  23.063}]```

I'm going to fiddle a bit more to see about adding support for setting names on the probes.  Worst case that can be done on the consuming side, so I won't spend a massive amount of time on it.

speshak, 2020-09-05:
After playing with my change a bit, I decided that it slows down the temp report too much (and understandably gets longer as more probes are added)

I added a new command, tentatively named 'T' to get the raw device readings.  This should minimize the impact my crazy feature has on other users.

speshak, 2020-09-30:
#46 Fixed this issue.

PR #46 (2020-09-21): Report all probes
Fixes #45
closed 2020-09-30 by speshak