thorrak/brewpi-esp #49: Improve code documentation

speshak, 2020-09-12:
Add more complete documentation for class/method level details.

(I'm opening this a bit as a forewarning.  I've been working on getting Doxygen comments into as much of the codebase as I can understand.  I don't want anyone to be blindsided by a PR that touches almost every file)

speshak, 2020-09-12:
My WIP branch: https://github.com/speshak/brewpi-esp8266/tree/cleanup_and_docs
I'm trying to keep this limited to just adding documentation, but some whitespace normalization has snuck in as well.

thorrak, 2020-09-12:
Works for me - Whenever you have some time, send through some PRs. There
are going to be a lot of files to review as you noted. :)

speshak, 2020-10-18:
I think the hard work for this is done.  There will be a long tail of adding & improving the documentation, but we're pointed in the right direction and this can be closed.

PR #51 (2020-09-13): Add Doxygen documentation
- Adds the comments to a large chunk of code
- Adds a GitHub action to automatically publish the docs to GitHub Pages
- Cleans up some code noticed while documenting.

I wouldn't say this Fixes #49, but it certainly moves the ball forward on it.
closed 2020-10-18 by speshak