thorrak/brewpi-esp #50: Fluctuation issues

dannyshaw, 2020-09-12:
Not sure whether this should be here or with Fermentrack, or if it's an issue at all.. 

I'm seeing a pattern for my setup where the fridge seems to kick in a little early and plunges the temp too low, resulting in a substantial dips and the heater needing to kick in to rectify it. I assumed there'd be some parameters I could tweak the algorithm somewhere? Is that a thing?

I should note the beer temp is a probe taped to the outside of a plastic tub with a patch of neoprene insulating it, perhaps a thermowell would provide a different experience? I havent got a fan going yet either. perhaps that would help.

Any feedback would be appreciated
 
[image]

lbussy, 2020-09-12:
Tuning a PID loop is not for the faint-hearted and rarely needed with these.  Why don't you tell us about the physical setup?  How big is hte fridge, how big the heater, what batch size, where are the probes?

dannyshaw, 2020-09-12:
Cheers, Sure thing.

- Westinghouse 240L, all fridge chamber. I've bypassed the thermostat with my controller.
- Heater is a small desk lamp down in the crisper well, with a 50W ceramic infrared bulb
- Test batch was just 23 Litres of water
- Beer probe was taped to the side of the fermenter with a neoprene beer cooler as an insulator (taped all around it).
- Fridge probe is about 1/3 from the top on the side.
- There's a small fan built into the top fridge, it kicks in when the condenser is on, i plan to override it to be on at all times.

This is not my fridge, but it's this model:

[image]

lbussy, 2020-09-12:
All seems pretty good.  I'd definitely use the fan constantly, that will help.  As I said, the beer temp looks good so the fluctuations themselves don't seem bad.  Add the fan and see what happens.

If you still get the whipsaw after adding the fan, bump up the cooling overshoot values.  I have NO idea where that is in Fermentack.  If you are using the controller in BrewPi (Remix) then it's in Maintenance Panel > Advanced Settings > Cooling overshoot estimator.   I've never had to mess with that so I don't have an idea for a starting point.  It *is* a self learning parameter, so you can see if you have an existing value there and then increase it a but and see what happens.

closed 2022-03-24 by thorrak