thorrak/brewpi-esp #63: ESP32 branch returns wrong temperature

PoulKlemmensen, 2021-04-21:
I have been trying to compile and use the latest ESP32 branch, and using the wifi option. But when trying to connect to fermentrack, I get an error. And as I can read it, the Arduino returns a wrong temperature format. But I don't have the knowledge to find this error. Returning to the master branch works perfect.

 Apr 20 2021 19:37:59   JSON decode error: Invalid control character at: line 1 column 62 (char 61)
 Apr 20 2021 19:37:59   Line received was: h:[{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":2,"d":false,"p":12,"v":"Â ?@","a":"28C952810B000096","j":" 0.000"},{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":2,"d":false,"p":12,"v":"Â ?Âº","a":"288BA8800B000019","j":" 0.000"},

thorrak, 2021-04-22:
The ESP32 branch isn't ready for prime time yet, and this is, indeed, a bug.

speshak, 2021-04-22:
I probably introduced this with my PiLink refactor (and related changes).  @PoulKlemmensen any chance you could get the hex values of that `v` value?  I'm not sure if it will help debug or not, but it might.

Could you also try connecting to your ESP via telnet and issue a `T` command?  That might pare down the number of places this bug lurks.

lbussy, 2021-04-22:
Pretty sure Thorrak disabled the Telnet capability when he moved to master.

PoulKlemmensen, 2021-04-22:
@speshak. Well, I guess that I can do that, but I probably need a few hints on how to do it. "Issue a T command" ?? Do I e.g. need to set a flag when compiling for getting debug flag return via the serial connection or ??

thorrak, 2021-04-22:
> Pretty sure Thorrak disabled the Telnet capability when he moved to master.

Wrong project. :)

lbussy, 2021-04-22:
> Wrong project. :)

Craaaaap.  I need to do fewer things at once.

PoulKlemmensen, 2021-04-22:
@speshak Thanks for the guide, I will load the firmware tonight and return the result.

PoulKlemmensen, 2021-04-22:
@speshak Think that these are the data you requested?

R:[{"device":"28FFC620C1170240","value":" 9.375","name":"28FFC620C1170240"},{"device":"28FF0938C11701CB","value":" 9.375","name":"28FF0938C11701CB"}]

T:{"BeerTemp":0,"BeerSet":20,"BeerAnn":null,"FridgeTemp":0,"FridgeSet":20,"FridgeAnn":null,"RoomTemp":"","State":1}

Error deserializing JSON data IncompleteInput
h:[{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":2,"d":false,"p":12,"a":"28FFC620C1170240","j":" 0.000"},{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":2,"d":false,"p":12,"a":"28FF0938C11701CB","j":" 0.000"},{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":false,"p":16,"x":true},{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":false,"p":14,"x":true},{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":false,"p":13,"x":true}]

PoulKlemmensen, 2021-04-22:
@speshak I will leave the firmware in for now, if there is additional.

PoulKlemmensen, 2021-04-23:
I have been looking at the above issue. And as far as I have gotten, the issue is in this module here:
void DeviceManager::serializeJsonDevice(JsonDocument& doc, device_slot_t slot, DeviceConfig& config, const char* value)

And if I change this line here, it kindof works:
	if (value && *value) {
//    deviceObj[DeviceDefinitionKeys::value] = value;
    deviceObj[DeviceDefinitionKeys::value] = Poul;	
	}

If looking in the log on fermentrack, I get something that looks fine:

 Apr 23 2021 09:43:17   Checking software version on controller... 
 Apr 23 2021 09:43:17   Found BrewPi v0.2.4, running commit 13805e1 build 13805e1, running on an ESP8266 ESP8266 with a DIY shield on port 192.168.100.10:23

 Apr 23 2021 09:43:17   BrewPi version received was 0.2.4 which this script supports in 'legacy' branch mode.
 Apr 23 2021 09:43:17   Bound to TCP socket on port 2110, interface localhost 
 Apr 23 2021 09:43:17   Installed devices received: null
 Apr 23 2021 09:43:22   Available devices received: [{"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": false, "p": 12, "v": 12.345, "a": "28FFC620C1170240", "j": " 0.000"}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 2, "d": false, "p": 12, "v": 12.345, "a": "28FF0938C11701CB", "j": " 0.000"}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": false, "p": 16, "x": true}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": false, "p": 14, "x": true}, {"i": -1, "t": 0, "c": 1, "b": 0, "f": 0, "h": 1, "d": false, "p": 13, "x": true}]

But now, I get an error when trying to assign control pins etc:

Request Method: | GET
-- | --
http://192.168.100.8/devices/5/sensors/
3.0.14
TypeError
'NoneType' object is not iterable
/home/fermentrack/fermentrack/app/models.py in load_all_from_devicelist, line 277
/home/fermentrack/venv/bin/python3.7
3.7.3
['/home/fermentrack/venv/bin',  '/home/fermentrack/venv/lib/python3.7/site-packages/git/ext/gitdb',  '/home/fermentrack/fermentrack',  '/home/fermentrack/venv/lib/python3.7/site-packages',  '/usr/lib/python37.zip',  '/usr/lib/python3.7',  '/usr/lib/python3.7/lib-dynload',  '.',  '/home/fermentrack/venv/lib/python3.7/site-packages/gitdb/ext/smmap']

thorrak, 2021-05-21:
I pushed a fix for this to the `lowdelay` branch (which is a fork of the ESP32 branch). The fix is tested on an ESP8266.

In addition to this change, I swapped the filesystem for ESP8266 to LittleFS. This change works on the ESP8266, but I may have neglected to fix some of the conditional imports for the ESP32 to ensure that it can continue to compile with SPIFFS. If anyone attempts to compile this branch for an ESP32 and runs into trouble, let me know and I'll see if I can figure out why.

PoulKlemmensen, 2021-06-07:
@thorrak Have you tried to change the control settings? When i am trying that, I get strange result back. Suspect that the change in temperature formats does something??

thorrak, 2021-06-07:
Yes, and it seemed to be working (using the lowdelay branch). What result do you get back?

PoulKlemmensen, 2021-06-08:
If I try and change e.g. the min, max and Kp, e.g. the three first parameters, they are 0'ed out and the Arduino returns 0. And I am using the latest lobdelay branch and compiling it myself. But I will try and debug later on today to see, that is happening.

And I have tried it out with your fermentrack. The reason for me saying that is, that I have made my own version of fermentrack, based on Node Red + Vue.js and something. But I have tried it with both versions and they react equal.

thorrak, 2021-06-09:
> If I try and change e.g. the min, max and Kp, e.g. the three first parameters, they are 0'ed out and the Arduino returns 0. And I am using the latest lobdelay branch and compiling it myself. But I [...]

Interesting -- I had issues with that up until ~3 weeks ago, but code in the lowdelay branch since then should have solved it. I'm not near any of my hardware at the moment or I would test it out, but I'll make note to try to test when I'm back with my test rigs.

PoulKlemmensen, 2021-06-11:
@thorrak. You probably know already, but it seems like this part does not work in my case:

The top part of
void SettingLoader::processSettingKeypair(JsonPair kv)

  if (kv.value().is<char *>()){
    str_value = kv.value().as<char *>();
  }
  else if (kv.value().is<float>()) {
    str_value = kv.value().as<float>();
  }
If e.g. I set the temperature, it is detected as char and is ok. In case of I altering the e.g. min and max temperatures, they are detected as floats and returns something funny.

thorrak, 2022-03-24:
After testing, this is now fixed in the experimental branch, and will be resolved when that branch is merged.

commit 112b5c6f (2021-05-21): Initialize device array (Fixes #63)
closed 2023-03-26 by thorrak