thorrak/brewpi-esp #67: esp32 branch git branch issue

bsaurusrex, 2021-06-18:
Hello,
  Any advice on how to fix this.

 - toolchain-xtensa 2.40802.200502 (4.8.2)
fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):
  File "C:\Users\brian\Desktop\BEER\BrewPi\brewpi-esp8266-esp32\scripts\gen_version.py", line 15, in <module>
    subprocess.check_output(["git", "describe", "--always"])
  File "C:\Users\brian\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\brian\.platformio\python3\lib\subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'describe', '--always']' returned non-zero exit status 128.
OSError: 'python3 scripts/gen_version.py' exited 1:
  File "C:\Users\brian\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\brian\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\brian\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\brian\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\brian\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\brian\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\brian\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\brian\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 124:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "C:\Users\brian\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\brian\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 222:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\brian\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\brian\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 185:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\brian\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Environment.py", line 821:
    do_parse(arg)
  File "C:\Users\brian\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Environment.py", line 673:
    arg = self.backtick(arg[1:])
  File "C:\Users\brian\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Environment.py", line 585:
    raise OSError("'%s' exited %d" % (command, status))

lbussy, 2021-06-18:
The first line tells you what's wrong. Did you clone the repository, or download as a zip file? There are tools which require and expect it to be a git repository.￼

bsaurusrex, 2021-08-01:
Thanks.  VScode/platformIO has a bit of a learning curve.  Running on LolinD32 tft nicely!

closed 2021-08-01 by bsaurusrex