thorrak/brewpi-esp #74: ESP32 Configuration Pin error when Not Inverted

Ap0ph1s-W4ite, 2022-11-06:
Configuring the Pins for ESP32 on Fermentrack gives an error when the option "Not Inverted" is selected.

ERROR:
> Failed to write the configuration to the controller. If this continues, try resetting the EEPROM on the controller.

If selected the "Inverted" option everything goes perfect and a succeed message appears. 

Maybe related to issue #62 because with my oscilloscope I see just before the error occurs the signal goes HIGH and the only way to getting it low is doing a RESET physically.

Resetting the EEPROM doesn't do nothing to this issue.

erviker, 2022-11-09:
Im having the same issue on ESP8266 

Firmware: brewpi-esp8266.lowdelay.wifi
Board NodeMCU ESP-8266
Firmware flashed using NodeMCU flasher

thorrak, 2023-03-01:
As it turns out, this is actually a bug in Fermentrack, but a fix can be applied in the firmware. I've added a fix in the v15-beta2 firmware as part of commit 629311f2fbf43040934129ce122374d7ae2cd9d1

thorrak, 2023-03-26:
Closed by #85

closed 2023-03-26 by thorrak