thorrak/brewpi-esp #75: no WIFI setup possible

bitzebraeu, 2023-02-01:


thorrak, 2023-02-01:
Try flashing the WiFi Reset & Sensor Wiring Test firmware. It's located under `Other > ESP8266` in BrewFlasher, or available [here](https://web.brewflasher.com/fw/8) on BrewFlasher Web.

Once you've run the test (or if you don't have a display connected, once you've flashed it, reset the device, and waited about two minutes) then re-flash the BrewPi-ESP8266 firmware and see if it works then.

One of the things the "test" firmware does is reset the controller to a consistent state which may help to correct any potential issues with the WiFi. Alternatively, you can also try the 0.14 beta version of the firmware, which has updated WiFi manager libraries.

bitzebraeu, 2023-02-01:
OK, will do. Just to be sure:
* test firmware: do I just flash it, or factorx reset my controller first and then flash?
* I tried the 0.14 already, insted of "ESD_..." in my network list I see "BrewPi" (or similar) and this requires a password in order to connect. Is there any?

bitzebraeu, 2023-02-01:
I explain step by step what I did:
* flashed the "test" fimware with BrewFlasher on my Windows PC
* reset and waited
* flashed the 0.11 WIFI version
* now hooked up the controller via USB to my PI where Fermentrack is installed
* connected to "ESP_..." WIFI
* tried to load 192.168.4.1  -->  nothing happened :-(

I assmue the 0.14 version will not bring a difference. However, what is the difference between "Low delay", "no spiffs" and the blank WIFI?

bitzebraeu, 2023-02-01:
after flashing the 0.14 version, the WIFI in my list is called "BrewPiAP" and requires a password. Which is it?

bitzebraeu, 2023-02-02:
I repeated the process about 5 times, did not change anything. Suddenly works. (v0.11)

closed 2023-02-02 by bitzebraeu

thorrak, 2023-02-02:
Weird. Well - if you ever want to use the v14 firmware, the ap password is “brewpiesp”

https://github.com/thorrak/brewpi-esp8266/blob/7930d1e2051f3f2511475d43cf78d32866b6a0f5/src/Config.h#L352

bitzebraeu, 2023-02-02:
Apart from the sp I guessed quite right :-) I'll give it a try once all is up and running..