thorrak/brewpi-esp #97: [Request] Tuya power outlet support

bsaurusrex, 2023-08-20:


thorrak, 2023-08-20:
Can you add links to these devices? (Even better - links to an API for them?)

bsaurusrex, 2023-08-20:
The devices are all generic but run under tuya cloud.  There is a lot of information available for home assistant addons like "localtuya" or using the traditional tuya cloud API.

https://vi.aliexpress.com/item/1005004562976013.html?spm=a2g0o.productlist.main.31.3399313dQDkOuq&algo_pvid=a2aaa5b0-46e6-409d-8df4-67b788402193&algo_exp_id=a2aaa5b0-46e6-409d-8df4-67b788402193-15&pdp_npi=4%40dis%21SGD%2154.61%2119.66%21%21%2139.10%21%21%402101f49f16925493100876351e3ad0%2112000029630940900%21sea%21SG%212655159861%21&curPageLogUid=NTv3yGJ4QAs0

bsaurusrex, 2023-08-20:
https://developer.tuya.com/en/docs/iot/call-api?id=Ka7o7ru1julms

thorrak, 2023-08-20:
I’m not likely to end up supporting the cloud, but if there is a (user friendly) way to support the devices locally I might be open. Is there any API documentation out there for local device support?

bsaurusrex, 2023-08-21:
Localtuya is an option.  https://github.com/rospogrigio/localtuya

bsaurusrex, 2023-10-24:
I picked up a Kasa outlet.

closed 2023-10-24 by bsaurusrex