thorrak/brewpi-esp #98: solderless - cant add kasa to actuator

bsaurusrex, 2023-09-06:
Hello,
 I just flashed a new lolin d32 with v15b wifi tft version and am trying to add a kasa kp105 as a cooling actuator but am receiving the following error:

```
Update Failed
Unable to connect to the controller or the update was rejected. Please check the information provided and reattempt to update.
```
I tried rebooting both devices without any improvement.

bsaurusrex, 2023-09-06:
I was able to tag my tilt as the beer temp sensor successfully.  Just the kasa switch wont assign.

thorrak, 2023-09-06:
The KP105 isn’t one of the devices I’ve tested against, and isn’t a device I currently support — but that doesn’t mean that it _can’t_ be supported necessarily!

If you have a computer with Python on it (a raspberry pi counts) do you mind running the following commands and seeing if this detects the switch:

```shell
pip install —upgrade python-kasa
kasa —type plug
``` 

…and seeing what it says? It will contain information like your location, so don’t paste the full result here, but what I want to know is if it detects the plug that you have. If it does, then try running the following and see if that switches the device on and off (note - either don’t plug something in to the Kasa when doing this and listen for the “click” or plug something that can tolerate being turned on and off (meaning a lamp, not a fridge!)):

```shell
kasa —type plug —host 192.168.X.X on
kasa —type plug —host 192.168.X.X off
``` 

Replace 192.168.X.X with the IP address listed for the plug in the earlier command. 

If it _doesn’t_ detect your plug then make sure the Pi and the Kasa device are on the same network.

bsaurusrex, 2023-09-06:
`kasa --type plug
No host name given, trying discovery..
Discovering devices on 255.255.255.255 for 3 seconds
`
Doesnt return anything.

Adding the IP work fine though.

```
kasa --type plug --host 192.168.1.111
== brewpikasa - KP105(UK) ==
	Host: 192.168.1.111
	Port: None
	Device state: True
	== Generic information ==
	Time:         2023-09-06 20:22:24 (tz: {'index': xxxx, 'err_code': 0}
	Hardware:     2.0
	Software:     1.0.0 Build 210118 Rel.165921
	MAC (rssi):   30:DE:xxxxxxx (-51)
	Location:     xxxxxxxxxxxx

	== Device specific information ==
	LED state: True
	On since: 2023-09-06 12:03:55

	== Modules ==
	+ <Module Schedule (schedule) for 192.168.1.111>
	+ <Module Usage (schedule) for 192.168.1.111>
	+ <Module Antitheft (anti_theft) for 192.168.1.111>
	+ <Module Time (time) for 192.168.1.111>
	+ <Module Cloud (cnCloud) for 192.168.1.111>
```

On and off work ok.

Inside brewpi, it is discovering it as a TPLink Switch type and the mac address.

thorrak, 2023-09-06:
That's good news! Definitely supportable, then, and the issue has to be something that is in BrewPi-ESP. If you use Google Chrome, do you mind attempting to assign the sensor once again, and copying/pasting the payload/response from Google Chrome's developer tools so that I can see exactly what is being sent to/received from the controller?

To do this, you will need to do the following:

1. Open up Chrome & navigate to the "Set Up Sensors/Actuators" page
2. **MacOS:** Go to the View menu > Developer > Developer Tools. **Windows:** Click the "Three Dots" menu next to your profile name > More tools > Developer tools. 
3. Click the "Network" tab on the "Developer Tool" pane
4. Attempt to assign the sensor
5. Wait for the "failure" message to pop up
6. Click on the "devices/" network call in the "Developer Tool" pane that was created when you assigned the sensor 
7. Copy the "Payload" and "Response" sub-pane contents and paste them here

Here's a 36 second video of the process on MacOS:
**https://www.youtube.com/watch?v=EaX20KH8o70**

bsaurusrex, 2023-09-06:
`{"c":1,"b":0,"f":3,"h":7,"p":0,"x":false,"d":false,"i":1,"a":"30:DE:4B:16:68:E7"}
`

`{"message":"Unable to process update"}`

thorrak, 2023-09-06:
I just pushed out what I _think_ should fix this issue as "v15c-beta1" in BrewFlasher. If you can take a look and let me know if that works, I would appreciate it!

bsaurusrex, 2023-09-07:
Thanks, I believe that worked as I heard a click when adding the Kasa device to the esp32.

thorrak, 2023-09-07:
Perfect! Reopening for now until I can get this merged into master, but for anyone who has the same issue, try out v15c.

thorrak, 2023-11-08:
v15d has been released which _should_ fix this.

closed 2023-11-08 by thorrak