thorrak/brewpi-esp #102: solderfree lolind32 control mode wont turn on

bsaurusrex, 2023-10-24:
Hello, 
  I have a lolin d32 with a kasa switch assigned, DS18b20 assigned to chamber temp, tilt assigned to beer temp.  For some reason I cant get control mode to turn on.  If I set fridge temp or set beer temp, nothing happens.

bsaurusrex, 2023-10-24:
No heat switch/relay is selected - temperature control will be cooling only.

This is the only error.[image]

thorrak, 2023-10-24:
Weird. Does the LCD display show both temps (the beer and the 'fridge')? What does it display when you try to set a fridge temp constant that is below the current temp (and therefore should toggle cooling)?

bsaurusrex, 2023-10-24:
Yup, I'm setting 0-5c and when I set mode the page just refreshes.

[image]

thorrak, 2023-10-24:
Ahhh. I'm assuming that if you give it a few seconds to a minute, the "Control Mode" doesn't change from "Off"?

mp-se, 2023-10-24:
Ive just updated my controller to s2 chips and i could not start them via the web ui, worked fine via fermentrack but i guess thats the serial interface. How do i get the logs out from the device? Any web serial or use the tx/rx pins?

thorrak, 2023-10-24:
@mp-se It’s almost guaranteed to be an issue with the web UI. If you check the JavaScript log on a desktop you should be able to see the error.

mp-se, 2023-10-24:
There is nothing in the developer console for javascript so no clues there. When I look in the network history there is only a lot of http get to the /lcd and /all_temp_control endpoints, should there not be a post for the temp control when setting the temperature ?

thorrak, 2023-10-24:
There definitely should be. I remember an issue that I thought I fixed regarding that in the UI codebase, but I may not have recompiled that into the released firmware. I’m traveling until Friday, but should be able to diagnose and fix then.

mp-se, 2023-10-24:
Ok, I flashed the 15c wifi via brewflasher

bsaurusrex, 2023-10-25:
Confirmed fermentrack is able to control it without issue.

mp-se, 2023-10-25:
I tried the v16 branch as well. same issue.

thorrak, 2023-10-27:
Yeah -- the v16 branch has a v16 UI that is itself on a separate branch that hasn't yet been compiled/dropped into the v16 repo, so the UI you're seeing is almost certainly the same as the one currently released. If it's a UI bug (which I'm almost certain it is, as I think I [caught & fixed it](https://github.com/thorrak/brewpi_esp_ui/commit/e6c604eb883e4d233d2de5f6277e5da75b50461a) in the v16 UI) then I should be able to fix it, recompile the UI gzip files, and build a new release of the SPIFFS partition for v15. I'll get a release going as soon as we get back home.

thorrak, 2023-10-27:
The UI repo is [now updated](https://github.com/thorrak/brewpi_esp_ui/commit/e6c604eb883e4d233d2de5f6277e5da75b50461a) for the v15 UI -- I'll build/test once we're back.

closed 2023-11-08 by thorrak

thorrak, 2023-11-08:
v15d has been released which _should_ fix this.

commit 7b713c96 (2023-11-08): Update UI (Fixes #102)
commit c41430d4 (2023-11-08): Update UI (Fixes #102)