thorrak/brewpi-esp #104: solderfree not seeing TP-Link sockets

ChuckGl, 2023-11-06:
Just did an install of vv15c Wifi TFT on a lolin d32 pro.  the brewpi gets on the network and I can access the webpage.  I have two TP-Link HS100 plugs that are on the network and reporting back when I run "kasa --type plug" from my Pi.  I do not see the plugs on in brewpi.  I tried both vv15c and vv15c-beta6.  I am also unable to set the control mode.  Not sure if I should be able with no actuators set.

thorrak, 2023-11-06:
The control mode bug is a known one that I need to push a release out to fix. Very soon, day job permitting. 

I’ll need to figure out the next steps for debugging when I am home and have access to a network with devices I can use to test. 

In the mean time, out of curiousity, what country are you in? The US? I ask, as I think some countries have restricted devices.

ChuckGl, 2023-11-06:
I am in Germany.  I can say I have used the HS100 plugs with several other projects.  Those are all on RPI, python based, and all use some method to directly interact with the plug.  The most recent one uses python-kasa to poll and control the plugs.

thorrak, 2023-11-07:
The UK is the one that I knew of which had separate concerns/requirements. When you scan with python-kasa, what model gets reported back? For me, for example, the HS103(US) in this part of the string:

`== Office Light Bar - HS103(US) ==`

thorrak, 2023-11-07:
Additionally, when you run `kasa  --host 192.168.5.119 sysinfo` (replacing the IP address with the device you are trying to use), what do you see next to `'mic_type':`? Do you see `IOT.SMARTPLUGSWITCH`?

ChuckGl, 2023-11-07:
Here is the output from both commands:

pi@raspberrypi:~ $ kasa
No host name given, trying discovery..
Discovering devices on 255.255.255.255 for 3 seconds
== PGPIO_3 - HS100(EU) ==
        Host: 192.168.2.10
        Device state: False
        == Generic information ==
        Time:         2023-11-07 07:05:54 (tz: {'err_code': 0, 'index': 41, 'zone_str': '(UTC+01:00) Amsterdam, Berlin, Bern, Rome, Stockholm, Vienna', 'tz_str': 'CET-1CEST,M3.5.0,M10.5.0/3', 'dst_offset': 60}
        Hardware:     1.0
        Software:     1.2.6 Build 200727 Rel.115547
        MAC (rssi):   B0:4E:26:64:2B:9A (-44)
        Location:     {'latitude': 50.083824, 'longitude': 8.240237}

        == Device specific information ==
        LED state: True
        On since: None

        == Modules ==
        + <Module Schedule (schedule) for 192.168.2.10>
        + <Module Usage (schedule) for 192.168.2.10>
        + <Module Antitheft (anti_theft) for 192.168.2.10>
        + <Module Time (time) for 192.168.2.10>
        + <Module Cloud (cnCloud) for 192.168.2.10>

== PGPIO_2 - HS100(EU) ==
        Host: 192.168.2.11
        Device state: False
        == Generic information ==
        Time:         2023-11-07 07:05:53 (tz: {'index': 41, 'err_code': 0}
        Hardware:     2.0
        Software:     1.5.6 Build 191125 Rel.083657
        MAC (rssi):   AC:84:C6:59:DF:D9 (-34)
        Location:     {'latitude': 50.0838, 'longitude': 8.2402}

        == Device specific information ==
        LED state: True
        On since: None

        == Modules ==
        + <Module Schedule (schedule) for 192.168.2.11>
        + <Module Usage (schedule) for 192.168.2.11>
        + <Module Antitheft (anti_theft) for 192.168.2.11>
        + <Module Time (time) for 192.168.2.11>
        + <Module Cloud (cnCloud) for 192.168.2.11>

pi@raspberrypi:~ $ kasa --host 192.168.2.10 sysinfo
No --type defined, discovering..
== System info ==
{'active_mode': 'none',
 'alias': 'PGPIO_3',
 'dev_name': 'Wi-Fi Smart Plug',
 'deviceId': '8006E117E7C0D34F5FB1ECF2857998231986A0D3',
 'err_code': 0,
 'feature': 'TIM',
 'fwId': '00000000000000000000000000000000',
 'hwId': '22603EA5E716DEAEA6642A30BE87AFCA',
 'hw_ver': '1.0',
 'icon_hash': '',
 'latitude': 50.083824,
 'led_off': 0,
 'longitude': 8.240237,
 'mac': 'B0:4E:26:64:2B:9A',
 'model': 'HS100(EU)',
 'oemId': '812A90EB2FCF306A993FAD8748024B07',
 'on_time': 0,
 'relay_state': 0,
 'rssi': -40,
 'sw_ver': '1.2.6 Build 200727 Rel.115547',
 'type': 'IOT.SMARTPLUGSWITCH',
 'updating': 0}
pi@raspberrypi:~ $

thorrak, 2023-11-07:
Thank you - this is perfect! I see the exact problem. Your switch reports its type in the `type` field rather than `mic_type` which means it fails the checks on these two lines:

https://github.com/thorrak/brewpi-esp8266/blob/54f8163d78b5daf0adc4c1adeec6a87dfb5b0fb8/src/tplink/TPLinkScanner.cpp#L76

https://github.com/thorrak/brewpi-esp8266/blob/54f8163d78b5daf0adc4c1adeec6a87dfb5b0fb8/src/tplink/TPLinkScanner.cpp#L84

The solution is just to check for either type or mic_type which is what it appears Python-kasa now does:

https://github.com/python-kasa/python-kasa/blob/bde07d117fcae6170bdb4d8b375f138aae6fde60/kasa/discover.py#L384

I’ll try to get this done tonight, along with merging in the UI fix for mode changes.

closed 2023-11-08 by thorrak

thorrak, 2023-11-08:
v15d has been released which _should_ fix this.

ChuckGl, 2023-11-08:
No luck.  Updated to BrewPi-ESP32 - vv15d WiFi TFT on the Lolin D32 Pro this morning.  I also pulled out a H110 (EU) I have to check it.  The BrewPi is still only reporting Pin 25, 26, 34. Part of other cleanup work I put a fresh Raspbian Bullseye and python-kasa on.  It is seeing all three Kasa with no changes to system beyond apt upgrade.  Let me know if there is anything I can pull or do to assist.  I gather you are in the US so direct testing with an EU model is out for you.  Here is the kasa output from all three units.

`root@raspberrypi:/home/pi# kasa
No host name given, trying discovery..
Discovering devices on 255.255.255.255 for 3 seconds
== PGPIO_3 - HS100(EU) ==
        Host: 192.168.2.10
        Port: 9999
        Device state: False
        == Generic information ==
        Time:         2023-11-08 08:01:31 (tz: {'err_code': 0, 'index': 41, 'zone_str': '(UTC+01:00) Amsterdam, Berlin, Bern, Rome, Stockholm, Vienna', 'tz_str': 'CET-1CEST,M3.5.0,M10.5.0/3', 'dst_offset': 60}
        Hardware:     1.0
        Software:     1.2.6 Build 200727 Rel.115547
        MAC (rssi):   B0:4E:26:64:2B:9A (-41)
        Location:     {'latitude': 50.083824, 'longitude': 8.240237}

        == Device specific information ==
        LED state: True
        On since: None

        == Modules ==
        + <Module Schedule (schedule) for 192.168.2.10>
        + <Module Usage (schedule) for 192.168.2.10>
        + <Module Antitheft (anti_theft) for 192.168.2.10>
        + <Module Time (time) for 192.168.2.10>
        + <Module Cloud (cnCloud) for 192.168.2.10>

== PGPIO_2 - HS100(EU) ==
        Host: 192.168.2.11
        Port: 9999
        Device state: False
        == Generic information ==
        Time:         2023-11-08 08:01:30 (tz: {'index': 41, 'err_code': 0}
        Hardware:     2.0
        Software:     1.5.6 Build 191125 Rel.083657
        MAC (rssi):   AC:84:C6:59:DF:D9 (-32)
        Location:     {'latitude': 50.0838, 'longitude': 8.2402}

        == Device specific information ==
        LED state: True
        On since: None

        == Modules ==
        + <Module Schedule (schedule) for 192.168.2.11>
        + <Module Usage (schedule) for 192.168.2.11>
        + <Module Antitheft (anti_theft) for 192.168.2.11>
        + <Module Time (time) for 192.168.2.11>
        + <Module Cloud (cnCloud) for 192.168.2.11>

== PGPIO_1 - HS110(EU) ==
        Host: 192.168.2.12
        Port: 9999
        Device state: False
        == Generic information ==
        Time:         2023-11-08 08:01:31 (tz: {'index': 41, 'err_code': 0}
        Hardware:     2.0
        Software:     1.5.6 Build 191125 Rel.083657
        MAC (rssi):   AC:84:C6:5D:99:74 (-49)
        Location:     {'latitude': 50.0838, 'longitude': 8.2402}

        == Device specific information ==
        LED state: True
        On since: None

        == Current State ==
        <EmeterStatus power=0.0 voltage=224.551 current=0.025 total=0.001>

        == Modules ==
        + <Module Schedule (schedule) for 192.168.2.12>
        + <Module Usage (schedule) for 192.168.2.12>
        + <Module Antitheft (anti_theft) for 192.168.2.12>
        + <Module Time (time) for 192.168.2.12>
        + <Module Cloud (cnCloud) for 192.168.2.12>
        + <Module Emeter (emeter) for 192.168.2.12>

Found 3 devices
root@raspberrypi:/home/pi# kasa --host 192.168.2.10 sysinfo
No --type defined, discovering..
== System info ==
{'active_mode': 'none',
 'alias': 'PGPIO_3',
 'dev_name': 'Wi-Fi Smart Plug',
 'deviceId': '8006E117E7C0D34F5FB1ECF2857998231986A0D3',
 'err_code': 0,
 'feature': 'TIM',
 'fwId': '00000000000000000000000000000000',
 'hwId': '22603EA5E716DEAEA6642A30BE87AFCA',
 'hw_ver': '1.0',
 'icon_hash': '',
 'latitude': 50.083824,
 'led_off': 0,
 'longitude': 8.240237,
 'mac': 'B0:4E:26:64:2B:9A',
 'model': 'HS100(EU)',
 'oemId': '812A90EB2FCF306A993FAD8748024B07',
 'on_time': 0,
 'relay_state': 0,
 'rssi': -39,
 'sw_ver': '1.2.6 Build 200727 Rel.115547',
 'type': 'IOT.SMARTPLUGSWITCH',
 'updating': 0}
root@raspberrypi:/home/toulouse# kasa --host 192.168.2.11 sysinfo
No --type defined, discovering..
== System info ==
{'active_mode': 'none',
 'alias': 'PGPIO_2',
 'dev_name': 'Smart Wi-Fi Plug',
 'deviceId': '8006B86064848C531F04E1ED80B678E21A053CB2',
 'err_code': 0,
 'feature': 'TIM',
 'fwId': '00000000000000000000000000000000',
 'hwId': 'A8D378F10FACF73C85F72B445B654AB1',
 'hw_ver': '2.0',
 'icon_hash': '',
 'latitude_i': 500838,
 'led_off': 0,
 'longitude_i': 82402,
 'mac': 'AC:84:C6:59:DF:D9',
 'model': 'HS100(EU)',
 'next_action': {'type': -1},
 'oemId': '0CC41572446631ACD46A8EDFC8942AE1',
 'on_time': 0,
 'relay_state': 0,
 'rssi': -32,
 'sw_ver': '1.5.6 Build 191125 Rel.083657',
 'type': 'IOT.SMARTPLUGSWITCH',
 'updating': 0}
root@raspberrypi:/home/toulouse# kasa --host 192.168.2.12 sysinfo
No --type defined, discovering..
== System info ==
{'active_mode': 'none',
 'alias': 'PGPIO_1',
 'dev_name': 'Smart Wi-Fi Plug With Energy Monitoring',
 'deviceId': '8006694BD2D4D2CEFF7B881C4FA6ECBC1A07E50B',
 'err_code': 0,
 'feature': 'TIM:ENE',
 'fwId': '00000000000000000000000000000000',
 'hwId': '044A516EE63C875F9458DA25C2CCC5A0',
 'hw_ver': '2.0',
 'icon_hash': '',
 'latitude_i': 500838,
 'led_off': 0,
 'longitude_i': 82402,
 'mac': 'AC:84:C6:5D:99:74',
 'model': 'HS110(EU)',
 'next_action': {'type': -1},
 'oemId': '1998A14DAA86E4E001FD7CAF42868B5E',
 'on_time': 0,
 'relay_state': 0,
 'rssi': -45,
 'sw_ver': '1.5.6 Build 191125 Rel.083657',
 'type': 'IOT.SMARTPLUGSWITCH',
 'updating': 0}`

thorrak, 2023-11-08:
Doh! My fault. I tested the wrong branch of code yesterday - the v15d release had some logical issues that prevented it from working.  

I just re-released a new v15d. When you have a moment, do you mind installing this new version and letting me know if it works for you?

ChuckGl, 2023-11-08:
Yep I can see them now.  I will try and do further testing later, but wanted to get you feedback right away.

commit 1c96bb5d (2023-11-08): Check for both type and mic type (Fixes #104)