thorrak/brewpi-esp #106: Connection to Wifi AP fails

hummerdash, 2024-01-26:
Hi,

I flashed the firmware to a D1 Mini (ESP8266 EX).
All fine.

The AP appears on my iPhone, Mac and iPad.

When I want to connect, it fails. everytime...

When I tried it with my laptop from work (Windows) it worked.

Why?

lbussy, 2024-01-26:
@thorrak I had this issue when I cobbled up the old WiFiManager.  There were some notes about needing to assign a channel and country to facilitate used by Apple OS's.

thorrak, 2024-01-26:
@hummerdash What country are you in, out of curiousity?

@lbussy That makes sense. Unfortunately:

https://github.com/thorrak/brewpi-esp8266/blob/6d422d683ba2ba6779f97a3ef7866b49ce6e6e4c/src/ESP_BP_WiFi.cpp#L119-L120

It's possible that everything is fixed in the latest Arduino Core 2.0, but I recall there being other issues with more recent cores as of the last time I tested things.

hummerdash, 2024-01-30:
@lbussy I am from Germany. I tried to change the channel to 3. no change. 
I did not try the "setCountry()" function...
Once I get new gear for my brewing setup, I will try. For now I am happy that everything works :-)

Thanks to all of you!

thorrak, 2024-01-30:
@hummerdash If you want to try, I included the fix @lbussy is suggesting in the v16-alpha0 release.

thorrak, 2024-08-23:
Other users seem to be having success here, so I'm going to tag this as pending pull. If someone has an issue in the future, I can reopen/investigate further.