thorrak/brewpi-esp #111: ESP8266 fails after a day or so

simonwhill, 2024-05-13:
Hi, I have 2 Fridges built on the thorrak PCB , both fail to respond to Ping or communicate with the docker container after a few days.
 
When reflashed with brewflasher they work again for a further few days but eventually fail again. 

I upgraded to the new fermentrack 2 v16 alpha0 firmware to see if it was a bug but it’s still disappearing. 

I have changed the 5v psu and the ESP8266, and all of the DS18B20 one wire sensors.

In the logs on the docker container, it just stops reporting in. 

Any advice or help greatly appreciated

Regards

Simon

thorrak, 2024-05-13:
Does the problem persist if you restart the controller without reflashing it? 

Can you get it to reconnect if you restart the Pi/device hosting Fermentrack without doing anything else?

Can you still access the on-device configuration web portal?

simonwhill, 2024-05-13:
Yes the problem does persist if reseting or power cycling, no the microcontroller does not respond again once the device fails it falls off the wifi network, once it fails its nolonger connected to the wifi so i cant connect to the web portal.

Thanks

Simon

thorrak, 2024-05-13:
That’s really strange. When it disconnects from WiFi, does it spin up the “setup” WiFi access point? Do you have a screen connected to the controller - and if so, does it continue to control temperatures?

simonwhill, 2024-05-14:
So I’ve just been to the brewery, and I checked them both, they are both offering to the BrewPiAP access point, once connected to them on my phone they both remember there mdns name and re connecting them to the local wifi they then reconnect to fermentrack.net and start logging again. the weird thing is they are within 2-3 meters of the access point, the ssid is only offering 2.4ghz. so I’m stumped, my third esp8266 is working just fine and does not disconnect.

simonwhill, 2024-05-16:
@thorrak is there a way to stop the esp8266 falling back to softap mode

simonwhill, 2024-05-21:
Interestingly the device seems to have forgotten the AP that’s been set, whereas when change the SSID it says that it can’t connect to the AP 
[image]

simonwhill, 2024-05-24:
To continue to add info to this thread, one of my fermenters fell off the wifi today, it’s a new build chiller but has the same issues as the others, the display sows its still working 
However on a power cycle it displays the connect to this ap page.
[image]

It does remember its MDNS Name and once configured to reconnect to wifi it resumes logging and remembers its connected one wire temp sensors 

[image]

[image]

Thanks, 
Simon

Spangen, 2024-06-11:
Hi;

To add to the information. I am experiencing the same problem. I run four fridges with heating/cooling in my basement. All run on BrewPi v0.2.4, running commit 6d422d6 build 6d422d6. Running on ESP8266 Wemos D1 Mini's. I  also have a 5th D1 running BrewPi in my office in 2nd floor for test purposes. The one in the office is connected to another AP, but connects to same SSID name (I run fastroaming with 802.11R on OpenWRT. Not that it should matter, as I don't think ESP8266 supports 802.11R). 

[image]

I experience the same disconection problem as described above. They disconnect, and start BrewPiAP. All loose SSID/Pass but remembers mDNS name. I have had 3 disconnects only last week. 

The strange thing I experience is that when they fall off wifi, all five controllers seems to fall off at exactly the same time. I can see that because all the brewlogs stops receiving data at same time. Even stranger is that the one in the office also falls off at same time, even on totally different AP and power supply.

I do not know if this might be related to the disconnects, but I want to add the following information: 

I am doing a LOT of experimenting with ESP8266/ESP32 these days. I was starting to wonder if the disconects somehow was related to when I unpacked/flashed another ESP. This is mainly based on the observation that the disconnects seem to coinside in time with when I am working on other ESP projects.

The world of ESP is new territory to me (I am old school Microchip PIC/assembler/Basic/electronics dude ;) ) As of now, I am only playing around in Arduino IDE, with 8266's and 32's so I have not gathered enough knowledge to validate if my suspition is valid, but I see there is a LOT going on "under the hood" in these chips/boards. Could there be some low level protocol "thingy" that the ESP's have way down in their code, that gets "triggered" wirelessly that makes this happen?  Like if "something" triggers an ESP-Now message for example, that makes other boards that receive it "clear something in flash"? 

This probably is far fetched, but when I compare the dropout times in the brewlog with the times I have unpacked a new board and flashed some Arduino sketch to it, they suspiciousely coinside. Also, the first months or so when i had the brewery up this spring, i did not have any dropouts at all. But then I was not playing with other ESP's ..

Just my two cents..

Anyway, thanks for great software, and greatings from Norway.

//Spangen

simonwhill, 2024-08-27:
Thats great thank you for persevering, can I ask how I will know when these changes are available?

Thanks

Simon

thorrak, 2024-08-27:
> Thats great thank you for persevering, can I ask how I will know when these changes are available?
> 
> Thanks
> 
> Simon

It's available now as v16-alpha1 in BrewFlasher and in [this](https://github.com/thorrak/brewpi-esp8266/releases/tag/v16-alpha1) prerelease on GitHub.

simonwhill, 2024-08-27:
I’ve seen similar behaviour with the esp32-s2, will the same fix be coming to that platform? or am i chasing a different fault

thorrak, 2024-08-27:
> I’ve seen similar behaviour with the esp32-s2, will the same fix be coming to that platform? or am i chasing a different fault

Good question - This isn't an issue that I have on my end, so I've largely been debugging by releasing potential fixes and waiting for someone to report back that it works. For the ESP8266 I had a number of people reach out -- but none for the S2.

I went ahead and just recompiled/released the S2 WiFi firmware including the same fix that was applied for the ESP8266 and uploaded it as v16-alpha1. If you want to give it a shot, let me know if you have any issues!

simonwhill, 2024-08-27:
thats great ill flash a couple and install on friday, thanks for yout hard work

Simon

simonwhill, 2024-08-28:
looks like the complie happened for ESP32 (there is a v16 alpha1 showing) but not ESP32-s2

[image]

simonwhill, 2024-08-28:
ok on further digging, my existing ESP32-S2 has been up for 20days and shows this fermware version even though it shows it as Alpha1 version has this version number at the end d0e5db5
[image]

But the most recent two ESP-S2 i have flashed both have this version No. 606ad7c

[image]