thorrak/brewpi-esp #113: is telnet removed on v16 alpha?

jalexandre0, 2024-06-23:
Hello there. 

After a long pause, I returned to homebrewing and took time to dust off my old ESP8266 controllers. 

I flashed one of them with v15, which detected the sensors and actuators I needed to put them to work. But, when I am in the control mode to set up my fridge, the screen goes grey and nothing happens.  I still can see the temps on telnet mode, but nothing changes. 

So, I used the BrewFlasher to flash 16, and the control mode in the interface works, but telnet ports are disabled, even if I select the brewpi-remix/fermentrack legacy option.

Is there a problem with telnet and v16 or I'm missing something?  I know fermentrack.net is great and will migrate to it when iSpindel support is ready. :)

thorrak, 2024-08-23:
Hey there - apologies for having missed this issue before!

No, telnet should not be disabled for v16 unless you have selected to sync with Fermentrack 2/Fermentrack.net.

thorrak, 2024-12-13:
Closing this, since I'm hoping that my comment resolved the question. If you still have issues here, let me know - happy to reopen and look at something!

closed 2024-12-13 by thorrak