thorrak/brewpi-esp #128: Screen rotation

SysAdminSmith, 2025-03-29:
Greetings:

I have everything up and running and am quite pleased (thank you for your hard work!).  My last issue is one of aesthetics, only.  For whatever reason, my screen output is portrait and not landscape.  Is this intentional?  If not, any solution to change this?  I notice I can rotate 180 (via the firmware web gui) and  that's it.    

Thank you!

thorrak, 2025-03-29:
Uhhhhhh…. Can you post a photo? Do you have a link to the LCD you’re using? 

That’s very unexpected.

SysAdminSmith, 2025-03-29:
https://www.amazon.com/dp/B0CKRJ81B5?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_1

And unfortunately I did not take a photo.

thorrak, 2025-03-29:
Ah. Is this the 4” (480x320) one specifically?

SysAdminSmith, 2025-03-29:
Yes (he said hesitantly).

-------- Original Message --------

thorrak, 2025-03-29:
So the user interface is only designed for the 240x320 screens, so I’m wondering if that may be part of why it appears rotated. Is it also only in the top half of the screen?

SysAdminSmith, 2025-03-29:
Correct

-------- Original Message --------

thorrak, 2025-03-30:
Well, on the one hand, you are already having more success than I would have predicted given the UI is designed for a different resolution, but I expect that is little comfort if it doesn’t look as you would want. Unfortunately, unless you want to [take a stab](https://github.com/thorrak/brewpi-esp8266/blob/master/src/DisplayTFT.cpp) at **scaling** the interface yourself I don’t think there’s a fix here. 

I’d be happy to add a configuration option for larger screens if you wanted to build it, but as I don’t have a screen of this resolution(and have too many hardware configurations to support as is!) this isn’t something I would be able to build on my end.